erwincoumans/tiny-differentiable-simulator #7: any example about the gradient calculation?

foolyc, 2020-05-26:
is there any demo about calculating the gradient of system output q, q_{dot} with respect to physical properties parameters(link length/ mass/ inertia...) in python?

erwincoumans, 2020-05-29:
Only the C++ has gradients, the Python bindings don't expose the gradients yet. It will be exposed.

erwincoumans, 2020-08-26:
Latest version has pytinydiffsim_dual using dual numbers (forward mode). 
There is a trivial example here:
https://github.com/google-research/tiny-differentiable-simulator/blob/master/python/examples/z_dual_numbers.py

The current code is also uploaded to pypi as revision 0.0.3, on Windows or Linux you can use 
~~~~~~~~~~~~~~~~
pip3 install pytinydiffsim --upgrade --user
~~~~~~~~~~~~~~~~
Will look into a more interesting example later.