erwincoumans/tiny-differentiable-simulator #17: undefined reference to `xg::newGuid()'

rainbow979, 2020-08-05:
I tried to run "make -j", but it has a error.

> `CMakeFiles/meshcat_zmq_example.dir/meshcat_zmq_example.o: In function `generate_uuid[abi:cxx11]()':
meshcat_zmq_example.cpp:(.text+0x13b): undefined reference to `xg::newGuid()'
collect2: error: ld returned 1 exit status
examples/CMakeFiles/meshcat_zmq_example.dir/build.make:105: recipe for target 'examples/meshcat_zmq_example' failed
make[2]: *** [examples/meshcat_zmq_example] Error 1`

rainbow979, 2020-08-05:
I tried to run cmake .., there are some warnings like

> USING third_party/zmq
CMake Warning at examples/CMakeLists.txt:66 (find_package):
  By not providing "Findnlohmann_json.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "nlohmann_json", but CMake did not find one.

  Could not find a package configuration file provided by "nlohmann_json"
  with any of the following names:

    nlohmann_jsonConfig.cmake
    nlohmann_json-config.cmake

  Add the installation prefix of "nlohmann_json" to CMAKE_PREFIX_PATH or set
  "nlohmann_json_DIR" to a directory containing one of the above files.  If
  "nlohmann_json" provides a separate development package or SDK, be sure it
  has been installed.

rainbow979, 2020-08-05:
ubuntu 18.04
cmake 3.18.1
g++ 7.5.0

erwincoumans, 2020-08-26:
For the latest version you should not need to install nlohmann, since it comes in the third_party library. In fact, you don't need to install any libraries now.
You can also install using pip3 install pytinydifsim, for the python bindings.

closed 2020-09-01 by erwincoumans

DuaneNielsen, 2020-09-12:
I think he was compiling the examples.  I believe there are still quite a few deps required for those.

erwincoumans, 2020-10-17:
Ah, good point. It is fixed in the refactor branch.
https://github.com/google-research/tiny-differentiable-simulator/tree/refactor
In third_party/crossguid modify the CMakeLists.txt as follows:
~~~~~~~~~~~~~~~~

ADD_LIBRARY(crossguid guid.cpp)

IF(WIN32)
        target_compile_definitions(crossguid PUBLIC GUID_WINDOWS PRIVATE GUID_WINDOWS)
ELSE()
IF(APPLE)
        target_compile_definitions(crossguid PUBLIC GUID_LIBUUID PRIVATE GUID_LIBUUID)
        target_link_libraries(crossguid PUBLIC uuid PRIVATE uuid)
ELSE()
        target_compile_definitions(crossguid PUBLIC GUID_LIBUUID PRIVATE GUID_LIBUUID)
        target_link_libraries(crossguid PUBLIC uuid PRIVATE uuid)

ENDIF()
ENDIF()

target_include_directories(crossguid PUBLIC . PRIVATE .)

~~~~~~~~~~~~~~~~

erwincoumans, 2020-12-13:
You may need
~~~~~~~~~~~
apt-get install uuid-dev
~~~~~~~~~~~