erwincoumans/tiny-differentiable-simulator #19: Issue during initial install of 'make -j'

BPDanek, 2020-08-28:
When trying to install the `make -j` step returns

Rerunning does the same.
```[ 22%] Built target pytinydiffsim
[ 26%] Built target opengl_imgui_test
[ 28%] Built target pendulum_example_gui
[ 31%] Built target tiny_urdf_parser_opengl_example
[ 35%] Built target opengl_test
[ 36%] Built target pytinyopengl3
[ 98%] Built target zmq
make: *** [all] Error 2

BPDanek, 2020-08-31:
It looks like there were issues during build too, got an error similar to: https://github.com/ros-planning/navigation/issues/617
resolved with: https://github.com/ros-planning/navigation/issues/617#issuecomment-391005166

erwincoumans, 2020-09-01:
Can't reproduce, it compiles/build, links and runs on Windows, Linux and Mac OSX.

closed 2020-09-01 by erwincoumans