erwincoumans/tiny-differentiable-simulator #20: Fmin is not a member of ceres

DuaneNielsen, 2020-09-10:
make

```
Scanning dependencies of target neural_contact
[ 75%] Building CXX object examples/CMakeFiles/neural_contact.dir/neural_contact.o
In file included from /home/duane/PycharmProjects/tiny-differentiable-simulator/examples/tiny_ceres_estimator.h:27:0,
                 from /home/duane/PycharmProjects/tiny-differentiable-simulator/examples/neural_contact.cpp:6:
/home/duane/PycharmProjects/tiny-differentiable-simulator/examples/ceres_utils.h: In static member function ‘static T CeresUtils<N>::min1(const T&, const T&)’:
/home/duane/PycharmProjects/tiny-differentiable-simulator/examples/ceres_utils.h:105:19: error: ‘fmin’ is not a member of ‘ceres’
     return ceres::fmin(a, b);
                   ^~~~
/home/duane/PycharmProjects/tiny-differentiable-simulator/examples/ceres_utils.h:105:19: note: suggested alternative: ‘asin’
     return ceres::fmin(a, b);
                   ^~~~
                   asin
/home/duane/PycharmProjects/tiny-differentiable-simulator/examples/ceres_utils.h: In static member function ‘static T CeresUtils<N>::max1(const T&, const T&)’:
/home/duane/PycharmProjects/tiny-differentiable-simulator/examples/ceres_utils.h:110:19: error: ‘fmax’ is not a member of ‘ceres’
     return ceres::fmax(a, b);
                   ^~~~
/home/duane/PycharmProjects/tiny-differentiable-simulator/examples/ceres_utils.h:110:19: note: suggested alternatives:
In file included from /home/duane/PycharmProjects/tiny-differentiable-simulator/examples/../tiny_neural_network.h:21:0,
                 from /home/duane/PycharmProjects/tiny-differentiable-simulator/examples/neural_scalar.h:8,
                 from /home/duane/PycharmProjects/tiny-differentiable-simulator/examples/neural_contact.cpp:3:
/usr/include/c++/7/cmath:1444:5: note:   ‘std::fmax’
     fmax(_Tp __x, _Up __y)
     ^~~~
In file included from /usr/include/features.h:424:0,
                 from /usr/include/x86_64-linux-gnu/c++/7/bits/os_defines.h:39,
                 from /usr/include/x86_64-linux-gnu/c++/7/bits/c++config.h:533,
                 from /usr/include/c++/7/iosfwd:38,
                 from /usr/include/c++/7/ios:38,
                 from /usr/include/c++/7/istream:38,
                 from /usr/include/c++/7/fstream:38,
                 from /home/duane/PycharmProjects/tiny-differentiable-simulator/examples/neural_contact.cpp:1:
/usr/include/x86_64-linux-gnu/bits/mathcalls.h:329:1: note:   ‘fmax’
 __MATHCALLX (fmax,, (_Mdouble_ __x, _Mdouble_ __y), (__const__));
 ^
examples/CMakeFiles/neural_contact.dir/build.make:62: recipe for target 'examples/CMakeFiles/neural_contact.dir/neural_contact.o' failed
make[2]: *** [examples/CMakeFiles/neural_contact.dir/neural_contact.o] Error 1
CMakeFiles/Makefile2:353: recipe for target 'examples/CMakeFiles/neural_contact.dir/all' failed
make[1]: *** [examples/CMakeFiles/neural_contact.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

fixed by adding -DCERES_USE_CXX11 to flags

```
vi tiny-differentiable-simulator/build/examples/CMakeFiles/neural_contact.dir/flags.make
```

```
# CMAKE generated file: DO NOT EDIT!
# Generated by "Unix Makefiles" Generator, CMake Version 3.10

# compile CXX with /usr/bin/c++
CXX_FLAGS =  -fPIC   -std=gnu++1z

CXX_DEFINES = -DBT_USE_DOUBLE_PRECISION -DCPPAD -DCPPADCG -DDYNAMIC_LOAD_X11_FUNCTIONS -DGFLAGS_IS_A_DLL=0 -DHPIPM -DSTATIC_LINK_SPD_PLUGIN -DUSE_EIGEN -DZMQ_STATIC -DCERES_USE_CXX11

CXX_INCLUDES = -I/home/duane/PycharmProjects/tiny-differentiable-simulator/third_party/eigen3 -I/home/duane/PycharmProjects/tiny-differentiable-simulator/examples/. -I/home/duane/PycharmProjects/tiny-differentiable-simulator/examples/.. -I/home/duane/PycharmProjects/tiny-differentiable-simulator/examples/../third_party/zeromq/include -I/usr/local/include/bullet -I/usr/include/python3.6m -I/usr/lib/llvm-7/include -isystem /usr/local/include -I/usr/include/eigen3 -I/home/duane/Downloads/libtorch/include -I/home/duane/Downloads/libtorch/include/torch/csrc/api/include -I/home/duane/PycharmProjects/tiny-differentiable-simulator/third_party/cpp_base64/include -I/home/duane/PycharmProjects/tiny-differentiable-simulator/third_party/tinyxml2/include
```

DuaneNielsen, 2020-09-11:
better fix...  added to /tiny-differentiable-simulator/examples/CMakeLists.txt

```
find_package(Ceres)
add_definitions(-DCERES_USE_CXX11)
```

erwincoumans, 2021-05-20:
We have a script to build Ceres, that should work out-of-the-box, using the Ceres in third_party (not system install!)
build_libs_unix.sh 
and
build_libs_windows.bat

closed 2021-05-20 by erwincoumans