erwincoumans/tiny-differentiable-simulator #33: apple clang 12 issues

meshula, 2020-10-31:
In order to compile under apple clang 12, I had to make a couple of tweaks to the code. Note that I didn't have time to get to the bottom of why the NeuralNetwork<MyAlgebra> py::class won't compile. It complains that the various functions, initialize, and so on can't be matched. Accordingly I commented it out for the purposes of getting everything else compiling and running.

The remainder of the diffs allow the library to built, and it runs properly, without the NeuralNetwork pybinding, of course.

````diff
diff --git a/python/pytinydiffsim.inl b/python/pytinydiffsim.inl
index 6ca0b04..accd6d9 100644
--- a/python/pytinydiffsim.inl
+++ b/python/pytinydiffsim.inl
@@ -327,6 +327,7 @@
   m.def("point_jacobian", &MyPointJacobian);
   m.def("inverse_kinematics", &MyInverseKinematics);
   
+  #if 0
   py::class_<NeuralNetwork<MyAlgebra>>(      m, "NeuralNetwork")
       .def(py::init<int, bool>())
       .def(py::init<int, const std::vector<int>&,NeuralNetworkActivation, bool >())
@@ -337,7 +338,7 @@
       .def("save_graphviz", &NeuralNetwork<MyAlgebra>::save_graphviz)
       ;
       //.def("compute_contacts", &CollisionDispatcher<MyAlgebra>::compute_contacts2);
-
+#endif
   py::class_<NeuralNetworkSpecification>(
       m, "NeuralNetworkSpecification")
       .def(py::init<int, bool>())
diff --git a/src/math/neural_network.hpp b/src/math/neural_network.hpp
index a05829e..b9c5513 100644
--- a/src/math/neural_network.hpp
+++ b/src/math/neural_network.hpp
@@ -259,7 +259,7 @@ class NeuralNetworkSpecification {
       file << "n_" << i - 1 << "_0--n_" << i << "_0;\n\t";
     }
     file << "edge[style=solid, tailport=e, headport=w];\n\t";
-    Algebra::Scalar max_weight = Algebra::zero();
+    typename Algebra::Scalar max_weight = Algebra::zero();
     if (!weights.empty()) {
       max_weight = (Algebra::abs(weights[0]));
       for (const auto& w : weights) {
diff --git a/src/tiny_inverse_kinematics.h b/src/tiny_inverse_kinematics.h
index 6b439d9..3328ddf 100644
--- a/src/tiny_inverse_kinematics.h
+++ b/src/tiny_inverse_kinematics.h
@@ -136,7 +136,7 @@ namespace TINY {
             VectorX& q) const {
 
             TinyIKResult<Scalar, Utils> result;
-            result.residual = -1;
+            result.residual = Scalar(-1);
 
             assert(q_init.size() == mb.dof());
             assert(q_reference.empty() || q_reference.size() == q_init.size());
@@ -208,10 +208,10 @@ namespace TINY {
                         JJTe_lambda2_I(i, i) += lambda * lambda;
                     }
                     Eigen::Matrix<double, Eigen::Dynamic, 1> eigen_mat =
-                        helper::to_eigen(JJTe_lambda2_I)
+                        to_eigen(JJTe_lambda2_I)
                         .colPivHouseholderQr()
-                        .solve(helper::to_eigen(e));
-                    VectorX z = helper::from_eigen_v<double, Utils>(eigen_mat);
+                        .solve(to_eigen(e));
+                    VectorX z = from_eigen_v<double, Utils>(eigen_mat);
                     delta_theta = J.mul_transpose(z);
                 }
 ````

erwincoumans, 2020-11-14:
Thanks for the report, it should be fixed now.

closed 2020-11-14 by erwincoumans

meshula, 2020-11-14:
Ah, C++17. Thanks!