erwincoumans/tiny-differentiable-simulator #91: tds joint issue

erwincoumans, 2021-02-23:
simulating floating base, or fixed base (with x,y,z translation + spherical joint, to mimic floating base), has some issue.
Attached is a repro case, and pybullet comparison.
[tds_joint_issues.zip](https://github.com/google-research/tiny-differentiable-simulator/files/6032243/tds_joint_issues.zip)

ErikGartner, 2021-02-24:
Hi Erwin!

I've noticed this issue as well, for example the using the [humanoid model](https://github.com/google-research/tiny-differentiable-simulator/blob/master/data/humanoid_updated_inertia.urdf) with a floating base very quickly accelerates until velocity blows up the simulation, but using a completely fixed base seems fine.

So far I've been unable to track down the issue. Do you have any guess to what may be the cause?

erwincoumans, 2021-02-24:
I first guessed it was due to floating base, but the issue also happens with fixed base + additional degrees of freedom to make an object move. It is also not specific to spherical joints. Eric Heiden is wrapping up a paper for a deadline end of this month, he will help tracking it down after that.

erwincoumans, 2021-04-12:
See description in this commit:
https://github.com/erwincoumans/tiny-differentiable-simulator/commit/288d822d84f6e3d0438b07740bf14ec64b0106e6

So for now, the workaround for this Issue #91 is either:

1) use a fixed base and manually add xyz trans and xyz rotational joints to the fixed base or
2) use a floating base, and make sure the base doesn't have any inertial offset. If there was an offset, move the collision and visual shape of the base in the opposite direction.

erwincoumans, 2021-04-18:
Although things have improved, there are still some failing cases, see attached zip file

1) floating base has an issue with mass matrix (not positive definite) at impact (line 48)

2) fixed base with xyz translational, xyz-rotational joint, similar issue

```
```
Assertion failed: is_positive_definite_b, file D:\dev\tds\tiny-differentiable-simulator\src\mb_constraint_solver.hpp, line 238
```
```

[tiny_urdf_parser_opengl_example.zip](https://github.com/google-research/tiny-differentiable-simulator/files/6332007/tiny_urdf_parser_opengl_example.zip)

eric-heiden, 2021-04-19:
#102 fixes the fixed-base ant leg contact simulation. The issue was in the URDF to multi-body conversion when the inertia matrix is rotated. Directly rotating the 3x3 inertia matrix is incorrect (can become asymmetric), instead I create a Transform with the defined rotation and then apply that to the RBI of the base and links.

The floating-base ant leg is still behaving incorrectly, I am looking into it.

ErikGartner, 2021-04-19:
Thanks! 
I tested #102 and it improves the simulation. :-) I'll continue experimenting with it and see how far I get.

erwincoumans, 2021-10-08:
Floating base should work reliably now, also with spherical joints. Make sure not to have any inertial center of mass offset at the base link.
Closing for now.