erwincoumans/tiny-differentiable-simulator #116: Python API segmentation fault

ErikGartner, 2021-05-20:
Using `world.get_mb_constraint_solver()` and `world.set_mb_constraint_solver()` will lead to segmentation fault due to double free from both world's destructor and PyBind11's destructor.

One solution is to remove them from the Python API until they are made safe.

erwincoumans, 2021-05-21:
Thanks for the report. Indeed, mixing memory allocations from Python as well as in C++ will easily lead to issues when using PyBind11. I'd like to remove the allocations in world and other locations, but it may require a bit of refactoring (breaking the API). Let's leave it open for now, until it is sorted.

erwincoumans, 2021-05-21:
One quick way would be to have the world a member of the default constraint solver, with a pointer to it, so it doesn't need to be deleted. Then we can get rid of the new in constructor and delete in the destructor.