erwincoumans/tiny-differentiable-simulator #121: The register_graphics_instance call needs one more input argument

h8907283, 2021-06-19:
I downloaded and installed TDS by doing `pip install -e .` in the root directory. I think `setup.py` ran successfully.

Then I tried to run `render_obj_gl.py` in the `python/examples` directory. It gave me the following error:

> X11 functions dynamically loaded using dlopen/dlsym OK!
X11 functions dynamically loaded using dlopen/dlsym OK!
Creating context
Created GL 3.3 context
Direct GLX rendering context obtained
Making context current
GL_VENDOR=Intel Open Source Technology Center
GL_RENDERER=Mesa DRI Intel(R) UHD Graphics (CML GT2)
GL_VERSION=4.6 (Core Profile) Mesa 20.0.8
GL_SHADING_LANGUAGE_VERSION=4.60
pthread_getconcurrency()=0
Version = 4.6 (Core Profile) Mesa 20.0.8
Vendor = Intel Open Source Technology Center
Renderer = Mesa DRI Intel(R) UHD Graphics (CML GT2)
Traceback (most recent call last):
  File "render_obj_gl.py", line 36, in <module>
    mesh = app.renderer.register_graphics_instance(shape, pos, orn, color, scaling, opacity)
TypeError: register_graphics_instance(): incompatible function arguments. The following argument types are supported:
    1. (self: pytinyopengl3.TinyGLInstancingRenderer, arg0: int, arg1: TINY::TinyVector3<float, TINY::FloatUtils>, arg2: TINY::TinyQuaternion<float, TINY::FloatUtils>, arg3: TINY::TinyVector3<float, TINY::FloatUtils>, arg4: TINY::TinyVector3<float, TINY::FloatUtils>, arg5: float, arg6: bool) -> int
Invoked with: <pytinyopengl3.TinyGLInstancingRenderer object at 0x7f43611bb0f0>, 1, [0.000000 0.000000 0.000000], [0.707107 0.000000 0.000000 0.707107], [1.000000 1.000000 1.000000], [0.100000 0.100000 0.100000], 0.4

So it looks like `register_graphics_instance` expects 7 input arguments instead of 6. The last input argument should be a `bool`. I searched online for what the last argument should be, `True` or `False`, but I couldn't find it. I entered either one of them. The example ran successfully, so I thought I would give you guys a head up. I'd love to know the correct value though, `True` or `False`.

BTW, `register_graphics_instance` is used in probably all of the python examples.

Thanks!

erwincoumans, 2021-06-24:
Thanks for the report, the examples/bindings need to be updated indeed. Will get back to this.

erwincoumans, 2021-09-23:
The render_obj_gl.py script is fixed, thanks for the report!

closed 2021-09-23 by erwincoumans