erwincoumans/tiny-differentiable-simulator #122: Use of inverse dynamics

TneitaP, 2021-06-21:
In the demo "laikago_footstep_ik", I find the process of :
- "forward kinematics"(**laikago_footstep_ik.cpp, Line412**);
- "inverse kinematics"(**laikago_footstep_ik.cpp, Line406~459**);
- "forward dynamic"(**laikago_footstep_ik.cpp, Line191**);

But do not find "inverse dynamic" of the MultiBody. However,  there exist corresponding function defined in  "src/daynamcs/**inverse_dynamics.hpp Line23**", there does not appear to be a call to this function in the provided example. 

Is that mean "inverse dynamic" has not been considered in this current **tds** framework?

erwincoumans, 2021-06-24:
Collision detection and response is used in various demos/examples.
We don't have examples of using inverse dynamics.

erwincoumans, 2022-03-15:
realistically, we only develop features that we need for our research, or accept contributions. So we don't develop features on request. So closing for now.

closed 2022-03-15 by erwincoumans