erwincoumans/tiny-differentiable-simulator #123: Spherical Joints at Contact, Mass Matrix

jyf588, 2021-06-29:
Hi all,

I created a simple humanoid model with 1 spherical joint and all other joints fixed. When I throw this humanoid to the floor, I got array out of bound [assertion failures](https://github.com/google-research/tiny-differentiable-simulator/blob/c5d7196387b709de60370254b4ded70a60d2d7b8/src/math/tiny/tiny_matrix6x3.h#L89) and simulation stopped without error messages.

After some digging, it appears that when constructing the mass matrix for contact resolution, the spherical joint case might indeed have some issues. I ended up changing [this line](https://github.com/google-research/tiny-differentiable-simulator/blob/master/src/dynamics/mass_matrix.hpp#L84) to `Algebra::assign_block(*M, Algebra::transpose(Fi), 0, qd_i, 6, 3);`, as the way assign_block is [implemented for 6*3 matrices](https://github.com/google-research/tiny-differentiable-simulator/blob/master/src/math/tiny/tiny_algebra.hpp#L365) seemed to cause out-of-bound errors due to the "fake transpose".

With this mod, the sim successfully runs, but the **energy does not seem to decrease** over time with zero restitution. Please see the video attached. In fact, if we wait long enough, the sim eventually explodes and gives NaNs.

I checked related issues #76 , #91 and made sure my model does not have inertial offset at root. The urdf model and python code is also attached.

[Attachments](https://drive.google.com/file/d/11v9N4l_Eq-CZJGCZGhNUMOSEuQwgf0Wl/view?usp=sharing)

Thanks for helping with this issue!
Yifeng

erwincoumans, 2021-08-25:
Thanks for the report, haven't had the time to check this out yet.

erwincoumans, 2021-10-06:
Thanks for the report. I had a look at your URDF and could reproduce the issue.
With some fixes, the URDF can be loaded in C++ code, modifying the tiny_urdf_parser_opengl_example.cpp (see attached zip).
Haven't tried your Python script yet.
Fixes are in here: https://github.com/google-research/tiny-differentiable-simulator/pull/16
[tiny_urdf_parser_opengl_example.zip](https://github.com/google-research/tiny-differentiable-simulator/files/7291352/tiny_urdf_parser_opengl_example.zip)
Here is a video, note that a small amount of joint damping is applied, otherwise the simulation becomes unstable (gains energy):
https://www.youtube.com/watch?v=mRfVjEAv538
PyBullet behaves similar, when the default link and joint damping is disabled. See attached PyBullet script.

[pybullet_check.zip](https://github.com/google-research/tiny-differentiable-simulator/files/7291373/pybullet_check.zip)