erwincoumans/tiny-differentiable-simulator #132: Docker build error

qodiuy, 2021-08-21:
I cannot create a docker image using the [Dockerfile](https://github.com/google-research/tiny-differentiable-simulator/blob/master/Dockerfile).
There is no python/requirements.txt file.
When I comment out this part(Line 111. `RUN pip3 install -r python/requirements.txt`) and run it again, I get the following error.
I would like to know the solution.

Environment
Ubuntu 18.04

Run command
```
$ docker build -t tiny_diff_sim . 
```
```
make[2]: *** [CMakeFiles/pagmo.dir/src/algorithms/nlopt.cpp.o] Error 1
CMakeFiles/pagmo.dir/build.make:81: recipe for target 'CMakeFiles/pagmo.dir/src/algorithms/nlopt.cpp.o' failed
make[1]: *** [CMakeFiles/pagmo.dir/all] Error 2
CMakeFiles/Makefile2:122: recipe for target 'CMakeFiles/pagmo.dir/all' failed
Makefile:159: recipe for target 'all' failed
make: *** [all] Error 2
The command '/bin/sh -c cmake -DCMAKE_BUILD_TYPE=Release . -DPAGMO_WITH_NLOPT=ON && cmake -DCMAKE_BUILD_TYPE=Release --build . . && cmake --build . . --target install' returned a non-zero code: 2
```

erwincoumans, 2021-08-25:
Probably best to disable/remove Pagmo if it breaks the build. It was just used for some experiment with CMA-ES, we can also do our own implementation.

erwincoumans, 2021-09-23:
We deleted the file, so this problem should be fixed now.

closed 2021-09-23 by erwincoumans