erwincoumans/tiny-differentiable-simulator #162: Can't seem to compile tds

garymcintire, 2021-10-02:
I've looked at this for a couple days and cannot seem to compile tds. The cmake looks good but it does say pthreads failed. The web seems to indicate this is not fatal.
I've attached the results of the  make  command... sorry, github won't allow me to attach so here it is

[  0%] Building CXX object rbdl/CMakeFiles/rbdl-static.dir/src/rbdl_version.cc.o
[  0%] Building CXX object rbdl/CMakeFiles/rbdl-static.dir/src/rbdl_mathutils.cc.o
[  1%] Building CXX object rbdl/CMakeFiles/rbdl-static.dir/src/rbdl_utils.cc.o
[  1%] Building CXX object rbdl/CMakeFiles/rbdl-static.dir/src/Constraints.cc.o
[  2%] Building CXX object rbdl/CMakeFiles/rbdl-static.dir/src/Dynamics.cc.o
[  2%] Building CXX object rbdl/CMakeFiles/rbdl-static.dir/src/Logging.cc.o
[  3%] Building CXX object rbdl/CMakeFiles/rbdl-static.dir/src/Joint.cc.o
[  3%] Building CXX object rbdl/CMakeFiles/rbdl-static.dir/src/Model.cc.o
[  3%] Building CXX object rbdl/CMakeFiles/rbdl-static.dir/src/Kinematics.cc.o
[  4%] Linking CXX static library librbdl-static.a
[  4%] Built target rbdl-static
[  5%] Building CXX object dear_imgui/CMakeFiles/dear_imgui.dir/imgui.cpp.o
[  5%] Building CXX object dear_imgui/CMakeFiles/dear_imgui.dir/imgui_demo.cpp.o
[  5%] Building CXX object dear_imgui/CMakeFiles/dear_imgui.dir/imgui_draw.cpp.o
[  6%] Building CXX object dear_imgui/CMakeFiles/dear_imgui.dir/imgui_widgets.cpp.o
[  6%] Building CXX object dear_imgui/CMakeFiles/dear_imgui.dir/imnodes.cpp.o
[  7%] Linking CXX shared library libdear_imgui.so
[  7%] Built target dear_imgui
[  7%] Building CXX object rbdl/CMakeFiles/rbdl-urdfreader.dir/addons/urdfreader/urdfreader.cc.o
[  8%] Building CXX object rbdl/CMakeFiles/rbdl-urdfreader.dir/addons/urdfreader/urdf/urdfdom/urdf_parser/src/check_urdf.cpp.o
[  8%] Building CXX object rbdl/CMakeFiles/rbdl-urdfreader.dir/addons/urdfreader/urdf/urdfdom/urdf_parser/src/pose.cpp.o
[  8%] Building CXX object rbdl/CMakeFiles/rbdl-urdfreader.dir/addons/urdfreader/urdf/urdfdom/urdf_parser/src/model.cpp.o
[  9%] Building CXX object rbdl/CMakeFiles/rbdl-urdfreader.dir/addons/urdfreader/urdf/urdfdom/urdf_parser/src/link.cpp.o
[  9%] Building CXX object rbdl/CMakeFiles/rbdl-urdfreader.dir/addons/urdfreader/urdf/urdfdom/urdf_parser/src/joint.cpp.o
[ 10%] Building CXX object rbdl/CMakeFiles/rbdl-urdfreader.dir/addons/urdfreader/tinyxml/tinystr.cpp.o
[ 10%] Building CXX object rbdl/CMakeFiles/rbdl-urdfreader.dir/addons/urdfreader/tinyxml/tinyxml.cpp.o
[ 10%] Building CXX object rbdl/CMakeFiles/rbdl-urdfreader.dir/addons/urdfreader/tinyxml/tinyxmlerror.cpp.o
[ 11%] Building CXX object rbdl/CMakeFiles/rbdl-urdfreader.dir/addons/urdfreader/tinyxml/tinyxmlparser.cpp.o
[ 11%] Building CXX object rbdl/CMakeFiles/rbdl-urdfreader.dir/addons/urdfreader/urdf/boost_replacement/printf_console.cpp.o
[ 12%] Building CXX object rbdl/CMakeFiles/rbdl-urdfreader.dir/addons/urdfreader/urdf/boost_replacement/string_split.cpp.o
[ 12%] Linking CXX static library librbdl-urdfreader.a
[ 12%] Built target rbdl-urdfreader
[ 12%] Building CXX object opengl/CMakeFiles/opengl_window.dir/tiny_x11_opengl_window.cpp.o
[ 12%] Building CXX object opengl/CMakeFiles/opengl_window.dir/tiny_camera.cpp.o
[ 13%] Building CXX object opengl/CMakeFiles/opengl_window.dir/tiny_egl_opengl_window.cpp.o
[ 13%] Building CXX object opengl/CMakeFiles/opengl_window.dir/tiny_font_stash.cpp.o
[ 14%] Building CXX object opengl/CMakeFiles/opengl_window.dir/tiny_fonts.cpp.o
[ 14%] Building CXX object opengl/CMakeFiles/opengl_window.dir/tiny_gl_instancing_renderer.cpp.o
[ 15%] Building CXX object opengl/CMakeFiles/opengl_window.dir/tiny_gl_primitive_renderer.cpp.o
[ 15%] Building CXX object opengl/CMakeFiles/opengl_window.dir/tiny_gl_render_to_texture.cpp.o
[ 15%] Building CXX object opengl/CMakeFiles/opengl_window.dir/tiny_glfw_opengl_window.cpp.o
[ 16%] Building CXX object opengl/CMakeFiles/opengl_window.dir/tiny_load_shader.cpp.o
[ 16%] Building CXX object opengl/CMakeFiles/opengl_window.dir/tiny_open_sans.cpp.o
[ 17%] Building CXX object opengl/CMakeFiles/opengl_window.dir/tiny_opengl3_app.cpp.o
[ 17%] Building CXX object opengl/CMakeFiles/opengl_window.dir/tiny_opengl_fontstashcallbacks.cpp.o
[ 17%] Building C object opengl/CMakeFiles/opengl_window.dir/__/__/__/third_party/glad/gl.c.o
[ 18%] Building C object opengl/CMakeFiles/opengl_window.dir/__/__/__/third_party/glad/glx.c.o
[ 18%] Building CXX object opengl/CMakeFiles/opengl_window.dir/__/__/__/third_party/tinyobjloader/tiny_obj_loader.cc.o
[ 19%] Building CXX object opengl/CMakeFiles/opengl_window.dir/utils/tiny_clock.cpp.o
[ 19%] Building CXX object opengl/CMakeFiles/opengl_window.dir/utils/tiny_chrome_trace_util.cpp.o
/home/gary/tds/src/visualizer/opengl/utils/tiny_chrome_trace_util.cpp: In member function ‘void TinyTimings::flush()’:
/home/gary/tds/src/visualizer/opengl/utils/tiny_chrome_trace_util.cpp:106:71: warning: format ‘%lu’ expects argument of type ‘long unsigned int’, but argument 4 has type ‘long long unsigned int’ [-Wformat=]
               threadId, startTimeDiv1000, startTimeRem1000Str, newname);
                                                                       ^
/home/gary/tds/src/visualizer/opengl/utils/tiny_chrome_trace_util.cpp:110:67: warning: format ‘%lu’ expects argument of type ‘long unsigned int’, but argument 4 has type ‘long long unsigned int’ [-Wformat=]
               threadId, endTimeDiv1000, endTimeRem1000Str, newname);
                                                                   ^
[ 19%] Building CXX object opengl/CMakeFiles/opengl_window.dir/utils/tiny_logging.cpp.o
[ 20%] Building CXX object opengl/CMakeFiles/opengl_window.dir/__/__/__/third_party/stb_image/stb_image_write.cpp.o
[ 20%] Building CXX object opengl/CMakeFiles/opengl_window.dir/__/__/__/third_party/stb_image/stb_image.cpp.o
[ 21%] Linking CXX shared library libopengl_window.so
[ 21%] Built target opengl_window
[ 22%] Building CXX object rbdl/CMakeFiles/rbdl_gui_example.dir/examples/rbdl_gui_example.cpp.o
In file included from /home/gary/tds/third_party/rbdl/../../src/visualizer/opengl/tiny_camera.h:18:0,
                 from /home/gary/tds/third_party/rbdl/../../src/visualizer/opengl/tiny_common_graphics_app_interface.h:18,
                 from /home/gary/tds/third_party/rbdl/../../src/visualizer/opengl/tiny_opengl3_app.h:18,
                 from /home/gary/tds/third_party/rbdl/examples/rbdl_gui_example.cpp:14:
[... 66 lines cut ...]
     if constexpr (std::is_same_v<AlgebraFrom, AlgebraTo>) {
        ^
/home/gary/tds/third_party/rbdl/../../src/utils/conversion.hpp:119:7: error: ‘else’ without a previous ‘if’
     } else {
       ^
In file included from /home/gary/tds/third_party/rbdl/../../src/contact_point.hpp:3:0,
                 from /home/gary/tds/third_party/rbdl/examples/rbdl_gui_example.cpp:23:
/home/gary/tds/third_party/rbdl/../../src/geometry.hpp: In member function ‘tds::Capsule<Algebra>::Scalar tds::Capsule<Algebra>::distance(const Vector3&) const’:
/home/gary/tds/third_party/rbdl/../../src/geometry.hpp:140:24: error: ‘clamp’ is not a member of ‘std’
                p.z() - std::clamp(p.z(), -this->length / Scalar(2),
                        ^
In file included from /home/gary/tds/third_party/rbdl/examples/rbdl_gui_example.cpp:23:0:
/home/gary/tds/third_party/rbdl/../../src/contact_point.hpp: In function ‘int tds::contact_sphere_sphere(const tds::Geometry<Algebra>*, const tds::Pose<AlgebraFrom>&, const tds::Geometry<Algebra>*, const tds::Pose<AlgebraFrom>&, std::vector<tds::ContactPoint<Algebra> >&)’:
/home/gary/tds/third_party/rbdl/../../src/contact_point.hpp:62:6: error: expected ‘(’ before ‘constexpr’
   if constexpr (is_cppad_scalar<Scalar>::value) {
      ^
/home/gary/tds/third_party/rbdl/../../src/contact_point.hpp:75:5: error: ‘else’ without a previous ‘if’
   } else {
     ^
rbdl/CMakeFiles/rbdl_gui_example.dir/build.make:75: recipe for target 'rbdl/CMakeFiles/rbdl_gui_example.dir/examples/rbdl_gui_example.cpp.o' failed
make[2]: *** [rbdl/CMakeFiles/rbdl_gui_example.dir/examples/rbdl_gui_example.cpp.o] Error 1
CMakeFiles/Makefile2:397: recipe for target 'rbdl/CMakeFiles/rbdl_gui_example.dir/all' failed
make[1]: *** [rbdl/CMakeFiles/rbdl_gui_example.dir/all] Error 2
Makefile:145: recipe for target 'all' failed
make: *** [all] Error 2

erwincoumans, 2021-10-06:
Which compiler (and version etc) are you using?
It compiles on Windows 10 using Visual Studio 2019 fine here (just git clone and running the build_libs_windows.bat script.

garymcintire, 2021-10-09:
Erwin,

Thanks for responding. I'm on ubuntu 16.04. Thinking that was the problem, I clean installed 20.04. I'm getting to about 90% now but can't seem to understand the error. I'll run it again and get the error(s) for you. Again, thanks for your help.