erwincoumans/tiny-differentiable-simulator #192: ModuleNotFoundError for pytinydiffsim_ad and AttributeError for pytinydiffsim when running Python code

mhr, 2021-12-13:
I thought I'd just gotten Python working after finding https://github.com/google-research/tiny-differentiable-simulator/issues/63. But I was mistaken. I've found other errors as I've played with the example files for Python. Apologies if you would prefer if I had created multiple issues, one for each error.

```
$ python cppad_example.py
Traceback (most recent call last):
  File "cppad_example.py", line 1, in <module>
    import pytinydiffsim_ad as pd
ModuleNotFoundError: No module named 'pytinydiffsim_ad'
```

```
$ python b_urdf_structs_example.py
Traceback (most recent call last):
  File "b_urdf_structs_example.py", line 47, in <module>
    sphere_mb.set_base_position(pd.Vector3(0.,0.,25.))
AttributeError: 'pytinydiffsim.TinyMultiBody' object has no attribute 'set_base_position'
```

But other examples do work:
```
$ python e_load_urdf.py
robot_name= plane

$ python billiard_optimization.py
sphere2red_path= /mnt/c/users/mretc/OneDrive/Documents/Grad School/thesis/motion/tiny-differentiable-simulator/data/sphere2red.urdf
```

mhr, 2021-12-14:
To solve `ModuleNotFoundError for pytinydiffsim_ad`, I did the following:

1) I had to amend the issue I linked to and add `SET_TARGET_PROPERTIES(pytinydiffsim_ad PROPERTIES PREFIX "")` to python/CMakeLists.txt.
2) I had to run `make` with the flag `-D USE_CPPAD=ON`.

Still not sure how to fix the other issue.

erwincoumans, 2022-02-07:
If you can import pytinydiffsim_dual , you should be fine. We haven't updated the setup.py code to include the pytinydiffsim_ad I think.

closed 2022-03-15 by erwincoumans