erwincoumans/tiny-differentiable-simulator #200: The effect of accelerations on the velocities is added twice

jotix16, 2022-04-14:
Considering the example in [ant_environment.h](https://github.com/erwincoumans/tiny-differentiable-simulator/blob/c1a9e06de33756bceacfaddeecc6bc4b6c98560c/examples/environments/ant_environment.h) or [laikago_environment.h](https://github.com/erwincoumans/tiny-differentiable-simulator/blob/c1a9e06de33756bceacfaddeecc6bc4b6c98560c/examples/environments/laikago_environment.h), the [Semi-Implicit-Euler integration](https://en.wikipedia.org/wiki/Semi-implicit_Euler_method) takes place in the following snippet

https://github.com/erwincoumans/tiny-differentiable-simulator/blob/c1a9e06de33756bceacfaddeecc6bc4b6c98560c/examples/environments/ant_environment.h#L196-L200

First `integrate_euler_qdd(*mb_, dt);` is called to add the effect of accelerations into change of velocities,
then `tds::integrate_euler(*mb_, dt);` is called to integrate once again.

But if we look into [integrator.hpp](https://github.com/erwincoumans/tiny-differentiable-simulator/blob/master/src/dynamics/integrator.hpp), we find the integration of acceleration on both functions, i.e in `integrate_euler(..)`:

https://github.com/erwincoumans/tiny-differentiable-simulator/blob/c1a9e06de33756bceacfaddeecc6bc4b6c98560c/src/dynamics/integrator.hpp#L95-L98

and in `integrate_euler_qdd(..)`
https://github.com/erwincoumans/tiny-differentiable-simulator/blob/c1a9e06de33756bceacfaddeecc6bc4b6c98560c/src/dynamics/integrator.hpp#L173-L175

Is this meant to be like this or is this an error and it should be taken care of?

erwincoumans, 2022-04-21:
The accelerations should be cleared to zero after 'integrate_euler_qdd', isn't it?
```
  Algebra::set_zero(mb.qdd());
```

jotix16, 2022-04-22:
Missed that one. Thank you, I am closing the issue.

closed 2022-04-22 by jotix16