erwincoumans/tiny-differentiable-simulator #205: Code Generation Tool Question

simon-lc, 2022-11-09:
Hello Erwin, 

This is an amazing package! I am just curious to know the code generation tool you used to generate dynamics models such as this one:
[https://github.com/erwincoumans/tiny-differentiable-simulator/blob/master/examples/environments/omp_model_ant_forward_zero.h](https://github.com/erwincoumans/tiny-differentiable-simulator/blob/master/examples/environments/omp_model_ant_forward_zero.h)

Thanks a lot for your help! 
Cheers, 
Simon

erwincoumans, 2022-12-05:
Thanks Simon. We used cppad-codegen to generate the code. You can try to use the build_libs* scripts to enable cppad and cppad-codegen.

closed 2022-12-05 by erwincoumans

simon-lc, 2022-12-12:
Thanks Erwin! This is super helpful!

Huzaifg, 2023-07-06:
Hi! This is a great package! 

I had a quick question related to this.
Do you have any example scrips showing how you generated these models with CppAD and CppADCodeGen? Looking through [differentiation.hpp](https://github.com/erwincoumans/tiny-differentiable-simulator/blob/master/src/utils/differentiation.hpp#L455), it seems like you provide functors  in the template argument when you call compile. I was wondering if you have any examples showing how this is used on a model level. 

Best
Huzaifa