erwincoumans/tiny-differentiable-simulator #210: Where can I get "cuda_model_laikago.dll" ?

hjyu94, 2023-03-13:
[image]

After running ars_train_policy_cuda project, I got this error.
Where can I get "cuda_model_laikago.dll" file?

erwincoumans, 2023-05-01:
It has been a while that I looked at that code. You need to build and run another executable (that uses CppAD/CppADCodeGen) that will generate the CUDA DLL, and then make sure that DLL is in the right location (I used symbolic links for that).

It is not user friendly. I highly recommend looking into NVIDIA Warp, Google Brax/Jax or OpenAL Triton instead (unless you have time digging into this C++ code).

hjyu94, 2023-05-02:
I truly appreciate your time and effort in helping me with this matter.
Your expertise and insights in this field have been invaluable to me, and I
cannot thank you enough for your continuous support.

Sincerely